leonvandenbeukel/7-Segment-Digital-Clock-V2 #2: Add 5 LED / Strip - 7 Segment Digital Clock

Eka1911, 2021-02-16:
Dear Leon,

I tried to make it by increasing the number of LEDs / Strips, from 3 LEDs to 5 LEDs.

But when you turn it on, many of the leds don't light up.
The following sketch that I changed.

Please provide solutions and directions, so that it can be perfect.

Thank you.

#include <Wire.h>
#include <RtcDS3231.h>                        // Include RTC library by Makuna: https://github.com/Makuna/Rtc
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <ESP8266HTTPUpdateServer.h>
#include <FastLED.h>
#include <FS.h>                               // Please read the instructions on http://arduino.esp8266.com/Arduino/versions/2.3.0/doc/filesystem.html#uploading-files-to-file-system
#define countof(a) (sizeof(a) / sizeof(a[0]))
#define NUM_LEDS 142                           // Total of 142 LED's    ////5 led per segment X 7 segment X 4 digit + 2 dot titik 
#define DATA_PIN D6                           // Change this if you are using another type of ESP board than a WeMos D1 Mini
#define MILLI_AMPS 4000                        // asumsi 800mA jika 1ed per segment maka 5 ledper segment 4.000mA
#define COUNTDOWN_OUTPUT D5                    // COUNTDOWN_OUTPUT D5 to buzzer 3 beeps when the countdown timer is reached

#define WIFIMODE 2                            // 0 = Only Soft Access Point, 1 = Only connect to local WiFi network with UN/PW, 2 = Both

#if defined(WIFIMODE) && (WIFIMODE == 0 || WIFIMODE == 2)
  const char* APssid = "7Segment-5Led";        
  const char* APpassword = "1234567890";  
#endif
  
#if defined(WIFIMODE) && (WIFIMODE == 1 || WIFIMODE == 2)
  #include "Credentials.h"                    // Create this file in the same directory as the .ino file and add your credentials (#define SID YOURSSID and on the second line #define PW YOURPASSWORD)
  const char *ssid = SID;
  const char *password = PW;
#endif

RtcDS3231<TwoWire> Rtc(Wire);
ESP8266WebServer server(80);
ESP8266HTTPUpdateServer httpUpdateServer;
CRGB LEDs[NUM_LEDS];

// Settings
unsigned long prevTime = 0;
byte r_val = 255;
byte g_val = 0;
byte b_val = 0;
bool dotsOn = true;
byte brightness = 255;
float temperatureCorrection = -3.0;
byte temperatureSymbol = 12;                  // 12=Celcius, 13=Fahrenheit check 'numbers'
byte clockMode = 0;                           // Clock modes: 0=Clock, 1=Countdown, 2=Temperature, 3=Scoreboard
unsigned long countdownMilliSeconds;
unsigned long endCountDownMillis;
byte hourFormat = 24;                         // Change this to 12 if you want default 12 hours format instead of 24               
CRGB countdownColor = CRGB::Green;
byte scoreboardLeft = 0;
byte scoreboardRight = 0;
CRGB scoreboardColorLeft = CRGB::Green;
CRGB scoreboardColorRight = CRGB::Red;
CRGB alternateColor = CRGB::Black; 
long numbers[] = {
  0b00000111111111111111111111111111111,  // [0] 0                  ////5 led per segment
  0b00000111110000000000000000000011111,  // [1] 1                  ////5 led per segment
  0b11111111111111100000111111111100000,  // [2] 2                  ////5 led per segment
  0b11111111111111100000000001111111111,  // [3] 3                  ////5 led per segment
  0b11111111110000011111000000000011111,  // [4] 4                  ////5 led per segment
  0b11111000001111111111000001111111111,  // [5] 5                  ////5 led per segment
  0b11111000001111111111111111111111111,  // [6] 6                  ////5 led per segment
  0b00000111111111100000000000000011111,  // [7] 7                  ////5 led per segment
[... 396 lines cut ...]
  byte sr1 = scoreboardRight / 10;
  byte sr2 = scoreboardRight % 10;

  displayNumber(sl1,3,scoreboardColorLeft);
  displayNumber(sl2,2,scoreboardColorLeft);
  displayNumber(sr1,1,scoreboardColorRight);
  displayNumber(sr2,0,scoreboardColorRight);
  hideDots();
}

void printDateTime(const RtcDateTime& dt)
{
    char datestring[20];

    snprintf_P(datestring, 
            countof(datestring),
            PSTR("%02u/%02u/%04u %02u:%02u:%02u"),
            dt.Month(),
            dt.Day(),
            dt.Year(),
            dt.Hour(),
            dt.Minute(),
            dt.Second() );
    Serial.println(datestring);
}

helpquick, 2021-02-16:
numbers is an array of type "long", which is limited to 32 bits, and with 5 LEDs, you'd need 35 bits.

try change to "long long"

niall-cooley, 2021-03-25:
@helpquick 
> numbers is an array of type "long", which is limited to 32 bits, and with 5 LEDs, you'd need 35 bits.
> 
> try change to "long long"

What if you want to reduct to 2 leds?

helpquick, 2021-03-25:
@niall-cooley 

> What if you want to reduct to 2 leds?

There's a number of places in the code which need to be updated if you change the number of LEDs, e.g. start indexes will all need to be updated, the pattern in the numbers[] array need to be updated, etc.

niall-cooley, 2021-03-25:
@helpquick  Yeah I was thinking that Ive updated the number of LED's and the indexes. Just trying to work out the pattern in numbers[] now :)

helpquick, 2021-03-25:
```
  _8 _9
_7     10
_6     11
  13 12
_5     _0
_4     _1
  _3 _2
```

your bit patterns would like like the above. Bit 0 - 1st LED, bit 13 = last LED

So, if yo put a 1 next to the LEDs which would be lit for a given number, you can put a 1 in the bit lit, and 0 in the unlit bit.

I worked backwards from bit 13, so you can type it in order. for a 1, it is 0 0 1 1 0 0 0 0 0 0 0 0 1 1

A quick go at it is below:

```
0x00111111111111  // 0
0x00110000000011  // 1
0x11111100111100  // 2
0x11111100001111  // 3
0x11110011000011  // 4
0x11001111001111  // 5
0x11001111111111  // 6
0x00111100000011  // 7
0x11111111111111  // 8
0x11111111001111  // 9
```

Or something like that.

niall-cooley, 2021-03-25:
@helpquick Thank you!

MacGyverr, 2021-04-08:
So what if you wanted to use 7 LED per segment, my issue is that long long won't work and you can't use double.

0b0000000111111111111111111111111111111111111111111,  //0
0b1111111000000000000000000000000000011111110000000,  //1
0b0000000111111111111110000000111111111111111111111,  //2
0b1111111111111100000000000000111111111111111111111,  //3
0b1111111000000000000001111111000000011111111111111,  //4
0b1111111111111100000001111111111111100000001111111,  //5
0b1111111111111111111111111111111111100000001111111,  //6
0b1111111000000000000000000000111111111111110000000,  //7
0b1111111111111111111111111111111111111111111111111,  //8
0b1111111000000000000001111111111111111111111111111,  //9
0b0000000000000000000000000000000000000000000000000,  //10 off
0b0000000000000000000001111111111111111111111111111,  //degress symbol
0b0000000111111111111111111111111111100000000000000,  //C
0b0000000000000011111111111111111111100000001111111,  //F

Any ideas?

niall-cooley, 2021-04-08:
Hi @MacGyverr im not 100% sure I would need to try it… You could possibly use a float?

helpquick, 2021-04-08:
@MacGyverr @niall-cooley 
You can't use float or double as the numbers array essentially holds a bitmap of which LEDs should be lit or not for each digit.

Why will ``long long`` not work? it should be 64 bits, which is enough to hold the required 49-bit bitmap? (I haven't actually tried more than 3 LEDs per segment).

niall-cooley, 2021-04-08:
> @MacGyverr @niall-cooley
> You can't use float or double as the numbers array essentially holds a bitmap of which LEDs should be lit or not for each digit.
>
> Why will `long long` not work? it should be [...]

@helpquick  your completely right I didn’t think on it like that! Just an update with help from your comments above I got 2 per segment working perfectly 😊
[image]

MacGyverr, 2021-04-08:
It starts to do weird things above 32 per segment, like never turning off certain lights and turning on less than the number it should, and if I use long long, the 32nd or 33rd light stays out all down the strip and it doesn't work right.
This array method looks promising:
byte digits[12][28] = {{0,0,0,0,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1},   // Digit 0
                       {0,0,0,0,1,1,1,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,1,1,1},   // Digit 1
                       {1,1,1,1,1,1,1,1,1,1,1,1,0,0,0,0,1,1,1,1,1,1,1,1,0,0,0,0},   // Digit 2
                       {1,1,1,1,1,1,1,1,1,1,1,1,0,0,0,0,0,0,0,0,1,1,1,1,1,1,1,1},   // Digit 3
                       {1,1,1,1,1,1,1,1,0,0,0,0,1,1,1,1,0,0,0,0,0,0,0,0,1,1,1,1},   // Digit 4
                       {1,1,1,1,0,0,0,0,1,1,1,1,1,1,1,1,0,0,0,0,1,1,1,1,1,1,1,1},   // Digit 5
                       {1,1,1,1,0,0,0,0,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1},   // Digit 6
                       {0,0,0,0,1,1,1,1,1,1,1,1,0,0,0,0,0,0,0,0,0,0,0,0,1,1,1,1},   // Digit 7
                       {1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1},   // Digit 8
                       {1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,0,0,0,1,1,1,1,1,1,1,1},   // Digit 9 | 2D Array for numbers on 7 segment
                       {1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,0,0,0,0,0,0,0,0,0,0,0},   // Digit *0
                       {0,0,0,0,0,0,0,0,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,0,0,0}};  // Digit C
and then
      for(int k=0; k<=27;k++){ 
        // Serial.print(digits[digit][k]);
        if (digits[digit][k]== 1){leds[cursor]=ledColor;}
         else if (digits[digit][k]==0){leds[cursor]=0x000000;};
         cursor ++;
        };
But my complete lack of programming skills and my inability to fix a simple size issue, makes me think I'll never be able to convert Leon's superior web-server based code into using that either. :)

helpquick, 2021-04-08:
Alternative option would be to have the numbers[] array be just 1 bit per segment, then add an extra for loop to expand it to the number of LEDs

```
// bitmap for LEDs 
byte numbers[] = { 
//  7654321
  0b0111111,  // [0] 0
  0b0100001,  // [1] 1
  0b1110110,  // [2] 2
  0b1110011,  // [3] 3
  0b1101001,  // [4] 4
  0b1011011,  // [5] 5
  0b1011111,  // [6] 6
  0b0110001,  // [7] 7
  0b1111111,  // [8] 8
  0b1111011,  // [9] 9
  0b0000000,  // [10] off
  0b1111000,  // [11] degrees symbol
  0b0011110,  // [12] C(elsius)
  0b1011100,  // [13] F(ahrenheit)
};
```

then:
 in the `displayNumber` function:

```
  for (byte i=0; i<7; i++){                // 7 segments
    for (byte j=0; j<7; j++ {              // 7 LEDs per segment
      yield();
      LEDs[j + i + startindex] = ((numbers[number] & 1 << i) == 1 << i) ? color : alternateColor;
    }
  } 
```

This is just an idea, I've not tested it.

MacGyverr, 2021-04-09:
That worked perfectly, I don't know why I didn't think to do that in the first place, it is so simple but so clever and best I can tell, doesn't cause any extra flickering or anything.
Thank you so much, you saved me soooooo much unnecessary work. :)

helpquick, 2021-04-16:
In case anyone is interested, I have uploaded an improved version of the code here: [7-Segment-WiFi-Clock](https://github.com/helpquick/7-Segment-WiFi-Clock)

I am still working on improvements, but as it stands, number of LEDs per segment can be configured, WiFi uses AutoConnect captive portal, so no need for hard-coded SSID / PSK, and time will update from NTP.