leonvandenbeukel/7-Segment-Digital-Clock-V2 #6: Blinking middle Dots

carl1961, 2021-05-09:
Any one have a clue where I can fine the code or a way to stop the blinking 2 dots? I looked every where in 7-Segment-WiFi-Clock.ino,  I have a hunch it may be somewhere in the FastLED lib files? Thanks for any advice

helpquick, 2021-05-09:
just comment out the ``displayDots`` on line 326.

carl1961, 2021-05-09:
> just comment out the `displayDots` on line 326 (in 7-Segment-Digital-Clock-V2/7-Segment-Clock/7-Segment-Clock.ino)

Thanks for the Fast post, Just tried that, Thanks, but it turns the dots off, I was trying to just stop the blinking.

helpquick, 2021-05-09:
@carl1961 
Ah, ok, replace the `displyDots` function (around line 403) with:

```
void displayDots(CRGB color) {
    LEDs[42] = color;
    LEDs[43] = color;
}
```

That'll for force the dots always on.

carl1961, 2021-05-09:
> @carl1961
> Ah, ok, replace the `displyDots` function (around line 403) with:
> 
> ```
> void displayDots(CRGB color) {
>     LEDs[42] = color;
>     LEDs[43] = color;
> }
> ```
> 
> That'll for force the dots always on.

Thanks @helpquick That worked Thanks! your true to your Name!

closed 2021-05-09 by carl1961