leonvandenbeukel/7-Segment-Digital-Clock-V2 #9: The first LED is an other color then the rest

Domo5785, 2021-12-18:
Hello,
my problem is that the first LED have an other Color then the rest

JBHTv, 2021-12-30:
I have the same problem

mhmhtssk, 2022-01-11:
Updating FastLed from the libraries section solves the problem. However, this time there is a control delay over wifi.

Protoncek, 2023-12-08:
I know it's old topic, but did you solve "first led problem"? i have this random color first led, too.  I converted my clock to additionally show seconds and  with 3.5.0 library it seems that whole clock runs slower -  two dots doesn't flash at all and seconds update only every two or three seconds... with 3.4.0 runs smoothly, but then first led is a problem... 

I'm thinking to insert one led at the front and hide it to eliminate this...