leonvandenbeukel/7-Segment-Digital-Clock-V2 #10: Schematic

Luki3012, 2022-01-08:
Hello,
Can you please put a schematic diagram online. I would like to recreate it.

Best wishes

JBHTv, 2022-01-22:
I used this diagram, but the clock does not show the time, only X0:00. countdown and score board works

[image]

Luki3012, 2022-01-22:
What is the black component on the left?

JBHTv, 2022-01-22:
a buzzer that works with the countdown

JBHTv, 2022-01-22:
https://es.aliexpress.com/item/32783807336.html
[image]

Luki3012, 2022-01-25:
Ok. Last question. Where did you get the lib "FastLED.h"?

JBHTv, 2022-01-26:
[image]
[image]

JBHTv, 2022-01-26:
So the connections have been left and the clock already works
[image]