leonvandenbeukel/7-Segment-Digital-Clock-V2 #15: CAN NOT GET THE CODE TO COMPILE Credentials.h

dmp32084, 2022-09-07:
I'm having a really hard time getting the code to compile in the Arduino program.
I am getting lost at the credentials.h   I made a file like your very bad and blunt instructions say to do but it just gets stuck there.
please help I have it all printed and  ready to go just cant get it to work. I have been working at this for weeks now and am getting very frustrated with it.

leonvandenbeukel, 2022-09-07:
The name of the file must start with an uppercase C (Credentials.h). Could that be the problem?

helpquick, 2022-09-07:
Actually, the readme is slightly incorrect. it says to create with UN and PW, but the code says:
```#include "Credentials.h"                    // Create this file in the same directory as the .ino file and add your credentials (#define SID YOURSSID and on the second line #define PW YOURPASSWORD)```

So, your Credentials.h should look like:
```
#define SID "yourssid"
#define PW "yourpassword"
```

Hope that helps.

If not, please post error messages.

dmp32084, 2022-09-08:
well now this is that I get when I get now

Arduino: 1.8.19 (Windows 10), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Disabled (new aborts on oom), Disabled, All SSL ciphers (most compatible), 32KB cache + 32KB IRAM (balanced), Use pgm_read macros for IRAM/PROGMEM, 4MB (FS:2MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\living room\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\living room\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Arduino wemos files\libraries -fqbn=esp8266:esp8266:d1_mini:xtal=80,vt=flash,exception=disabled,stacksmash=disabled,ssl=all,mmu=3232,non32xfer=fast,eesz=4M2M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600 -ide-version=10819 -build-path C:\Users\LIVING~1\AppData\Local\Temp\arduino_build_62009 -warnings=none -build-cache C:\Users\LIVING~1\AppData\Local\Temp\arduino_cache_147851 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mklittlefs.path=C:\Users\living room\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\3.0.4-gcc10.3-1757bed -prefs=runtime.tools.mklittlefs-3.0.4-gcc10.3-1757bed.path=C:\Users\living room\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\3.0.4-gcc10.3-1757bed -prefs=runtime.tools.mkspiffs.path=C:\Users\living room\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\3.0.4-gcc10.3-1757bed -prefs=runtime.tools.mkspiffs-3.0.4-gcc10.3-1757bed.path=C:\Users\living room\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\3.0.4-gcc10.3-1757bed -prefs=runtime.tools.python3.path=C:\Users\living room\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\living room\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\living room\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed -prefs=runtime.tools.xtensa-lx106-elf-gcc-3.0.4-gcc10.3-1757bed.path=C:\Users\living room\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed -verbose C:\Arduino\7-Segment-Clock\7-Segment-Clock.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\living room\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\living room\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Arduino wemos files\libraries -fqbn=esp8266:esp8266:d1_mini:xtal=80,vt=flash,exception=disabled,stacksmash=disabled,ssl=all,mmu=3232,non32xfer=fast,eesz=4M2M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600 -ide-version=10819 -build-path C:\Users\LIVING~1\AppData\Local\Temp\arduino_build_62009 -warnings=none -build-cache C:\Users\LIVING~1\AppData\Local\Temp\arduino_cache_147851 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mklittlefs.path=C:\Users\living room\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\3.0.4-gcc10.3-1757bed -prefs=runtime.tools.mklittlefs-3.0.4-gcc10.3-1757bed.path=C:\Users\living room\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\3.0.4-gcc10.3-1757bed -prefs=runtime.tools.mkspiffs.path=C:\Users\living room\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\3.0.4-gcc10.3-1757bed -prefs=runtime.tools.mkspiffs-3.0.4-gcc10.3-1757bed.path=C:\Users\living room\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\3.0.4-gcc10.3-1757bed -prefs=runtime.tools.python3.path=C:\Users\living room\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\living room\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\living room\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed -prefs=runtime.tools.xtensa-lx106-elf-gcc-3.0.4-gcc10.3-1757bed.path=C:\Users\living room\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed -verbose C:\Arduino\7-Segment-Clock\7-Segment-Clock.ino

Using board 'd1_mini' from platform in folder: C:\Users\living room\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2

Using core 'esp8266' from platform in folder: C:\Users\living room\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2

Detecting libraries used...

"C:\\Users\\living room\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\3.0.4-gcc10.3-1757bed/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -D_GNU_SOURCE "-IC:\\Users\\living room\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.2/tools/sdk/include" "-IC:\\Users\\living room\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.2/tools/sdk/lwip2/include" "-IC:\\Users\\living room\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\LIVING~1\\AppData\\Local\\Temp\\arduino_build_62009/core" -c - [... cut ...]
Using library ESP8266WebServer at version 1.0 in folder: C:\Users\living room\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WebServer 

Using library ESP8266HTTPUpdateServer at version 1.0 in folder: C:\Users\living room\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266HTTPUpdateServer 

Using library FastLED at version 3.5.0 in folder: C:\Arduino wemos files\libraries\FastLED 

exit status 1

expected primary-expression before 'const'

helpquick, 2022-09-08:
What does your Credentials.h look like?

dmp32084, 2022-09-08:
this is my Credentials.h

#define SID "Linksys"
#define PW "mypassword"
(mypassword is a sub so you don't see my password lol )

helpquick, 2022-09-09:
Strange, the only other idea i have is that you have accidentally made a change to the code in the .ino file, somwhere before line 24, or you have a corruption in your libraries / ardiono installation.

I tried to replicate your problem, but it compiles fine for me without errors. I even tried with some common mistakes in the Credentials.h, etc. but still compiles fine for me.

I'd try re-download the code into a new directory, create your Credentials.h and try compile again.

Make sure all libraries are up to date, make sure arduino software is up to date, etc.

Uninstall and reinstall arduino and libraries.